ra3xdh/qucs_s #1193: XSPICE Digital_AUX Library preliminary work

tomhajjar, 2025-01-06:
I started preliminary work on the Digital_AUX.lib to determine what will be in the Library.

As an experiment I made the Library knowing it would be missing most of the Spice models to see what it would look like. Since multiple Qucs-S Libraries like Xanalogue.lib also have Parameter-Passing and were made by hand I figured I might be able to add the missing info. I was surprised to see the following Verilog sections since they don't exist in any Qucs Library I have seen. Can you tell me what in my subcircuits causes Qucs "Create Library..." to add these entries?

******************************************************************************************************
  </VerilogModel>
  <VHDLModel>
library ieee;
use ieee.std_logic_1164.all;
entity Sub_Digital_AUX_ADC_bridge is
 generic (in_low : real := 0.5;
 in_high : real := 0.5;
 rise_delay : real := 1e-9;
 fall_delay : real := 1e-9;
 );
 port (net_net0 : inout  ;
 net_net1 : inout  );
end entity;
use work.all;
architecture Arch_Sub_Digital_AUX_ADC_bridge of Sub_Digital_AUX_ADC_bridge is
begin
end architecture;
  </VHDLModel>
******************************************************************************************************

[Digital_AUX_lib_prj.zip](https://github.com/user-attachments/files/18320854/Digital_AUX_lib_prj.zip)

ra3xdh, 2025-01-06:
> Can you tell me what in my subcircuits causes Qucs "Create Library..." to add these entries?

I suspect that some devices used in subcircuits contain pure digital model (Verilog/VHDL). There is no way to supress it inclusion in the library if such model is found.

tomhajjar, 2025-01-06:
I can't find any Library in Qucs, QucsStudio or Qucs-S that used "Verilog/VHDL". Seems like a lot of work for little return.

QucsStudio does have "Compiled Models" but not many use that feature.

tomhajjar, 2025-01-06:
Any idea if this will hinder adding Spice Library device with parameter passing to Create Library?

tomhajjar, 2025-01-06:
@andresmmera

Since you make Libraries by hand, any hints how to make a Library of this project?

andresmmera, 2025-01-06:
I'll take a look as soon as I can

tomhajjar, 2025-01-06:
Mike Brinson made Xanalogue.lib. Some of the subcircuits in this project should have similar structure.

tomhajjar, 2025-01-07:
Goals:
Libraries with XSPICE devices need parameter passing or they are useless. Parameter passing would be useful for other Libraries as well. We need to be able to make Libraries supporting parameter passing without resorting to making them manually like Xanalogue.lib. Presently only subcircuits made with the "Spice library device" support parameter passing but they cannot be made into a library using Qucs Create Library utility.

I made a simplified test project with XSPICE ADC & DAC Bridges, digital Clock and Analog Clock. I made two subcircuits each of the ADC & DAC Bridges using "Spice library" and "Spice netlist". Subcircuits made using Spice library support parameter passing and subcircuits made using Spice netlist do not.

Unfortunately the Qucs Create Library utility supports making a Library from subcircuits made using Spice netlist but doesn't support making one from subcircuits made using Spice library. I decided to make two Libraries "Digital_slib.lib" and "Digital_snet.lib" to see what the syntax differences looked like. 

As expected the devices in "Digital_snet.lib" "function" but don't support parameter passing. The devices in "Digital_slib.lib" don't work.  

In the case of subcircuits made with "digital" devices, Qucs Create Library inserts unrelated Verilog/VHDL sections into the Library cluttering it with spurious entries. 

[image]
[image]

[Digital_AUX_lib_test_prj.zip](https://github.com/user-attachments/files/18336175/Digital_AUX_lib_test_prj.zip)

ra3xdh, 2025-01-07:
> Parameter passing would be useful for other Libraries as well. 

Probably I will implement a way to pass parameters to *SPICE file* device. This task could be done before the release. Otherwise its need to wait #944 in the next release cycle. 

>  Verilog/VHDL sections into the Library cluttering it with spurious entries.

The Verilog/VHDL section should not affect the analog simulation.

tomhajjar, 2025-01-07:
When comparing "Digital_snet.lib" to "Xanalogue.lib" there are numerous extra entries that don't seem to do anything like "ModelIncludes "ADC_bridge.cir.lst" and "ModelIncludes "DAC_bridge.cir.lst" which are temp files never used again.

Digital_snet.lib also has a lot more entries than Xanalogue.lib even though both are for simple XSPICE devices.

ra3xdh, 2025-01-07:
> there are numerous extra entries that don't seem to do anything like "ModelIncludes "ADC_bridge.cir.lst" and "ModelIncludes "DAC_bridge.cir.lst" 

LST files are for Qucsator. These directives has no effect in SPICE simulation.

tomhajjar, 2025-01-07:
LST files are never used after the Library is made. They also don't exist on anyone's PC other than the original Author. I delete them all the time.

ra3xdh, 2025-01-08:
@tomhajjar The PR #1197 allows parameter passing for *SPICE file* component. You may proceed with the library design.

tomhajjar, 2025-01-08:
Great work Vadim. First test Library made is successful. Only a 100 devices left to convert...

Just have to remember copying the parameter info from the Spice library device before replacing it. The symbol and parameter info+data is not effected.

[image]
[image]

[Digital_AUX_lib_test_prj.zip](https://github.com/user-attachments/files/18347704/Digital_AUX_lib_test_prj.zip)

tomhajjar, 2025-01-08:
One nice thing about Spice library is you can copy-paste parameter info from the Spice file into the parameter field.

tomhajjar, 2025-01-08:
First cut at Digital_AUX.lib. I deleted a number of sources and pull-up/downs I deemed redundant.

One minor issue is sometimes a Digital simulation doesn't need a "ground" so instead of adding a resistor to ground I "grounded ground"...

[image]

[Digital_AUX_lib_prj.zip](https://github.com/user-attachments/files/18350682/Digital_AUX_lib_prj.zip)

tomhajjar, 2025-01-08:
My Internet is down. I have a newer version of the Lib plus more

ra3xdh, 2025-01-09:
> First cut at Digital_AUX.lib. 

Is this library ready for upload in the repository?

tomhajjar, 2025-01-09:
Yes. 

Also the goal is to also have all the files used to make the Libraries be available in the Discussion section. It should help spawn expansion of the Libraries and new ones. 

Having had to "reverse engineer" all the original "Qucs" Libraries isn't fun...

ra3xdh, 2025-01-09:
Library uploaded by #1200