ra3xdh/qucs_s #1195: Problem to specify log sweep

dwarning, 2025-01-07:
Made yesterday a compile from current branch with mingw-ucrt64 and have now problem to specify a log parameter sweep. The widget did not calculate the number of total points anymore. Simulation fails.

[image]

ra3xdh, 2025-01-07:
According the chapter 11.3.2 of the Ngspice manual the log sweep typ is not supported for DC simulation. The DC+ParSweep is converted to SPICE DC sweep. You cannot define log sweep for it.

ra3xdh, 2025-01-07:
This simulation type will work only for Qucsator. Something is wrong with points calculation. When I input 10 point per decade the total points gives 1. This may appear after #1054 @iwbnwif , Please have a look. 

[image]

dwarning, 2025-01-07:
No, with your provided executable 24.4.1 it works very well, as all the time:

[image]

ra3xdh, 2025-01-07:
To be honestly I cannot remember when I used the LOG Parameter Sweep the last time and if it was operational before.

ra3xdh, 2025-01-07:
> No, with your provided executable 24.4.1 it works very well, as all the time:

It works, but the result is totally wrong. The step is linear. The DC simulation type supports only linear step. It generates the following netlist line:

~~~
dc i1 1e-09 0.1 0.000833333
~~~

dwarning, 2025-01-07:
OK, now I see. My thought was initial that qucs-s transfer this to a pointwise simulation.
Correct would be that log disappears if I specify DC1 simulation type.
How the step 0.000833333 is calculated?

ra3xdh, 2025-01-07:
> How the step 0.000833333 is calculated?

~~~
step=(stop-start)/number
~~~

For my example (100m-1n)/121. This is not possible to fix, because DC doesn't support LOG sweep like AC.

dwarning, 2025-01-07:
I tried List with the provided binary.
[image]

This goes total wrong:

dc i1 0 2.63144e-312 1.17439e-311

In the end only lin sweep should be allowed and enumerated for ngspice active.

ra3xdh, 2025-01-07:
I have identified the root of the wrong calculation. The error is gone after the following patch:
~~~
diff --git a/qucs/components/componentdialog.cpp b/qucs/components/componentdialog.cpp
index 4f94f71e..92397082 100644
--- a/qucs/components/componentdialog.cpp
+++ b/qucs/components/componentdialog.cpp
@@ -597,13 +597,13 @@ void ComponentDialog::updateSweepProperty(const QString& property)
       if (property == "Start" || property == "Stop" || property == "Points" || property == "All")
       {
         double points = str2num(pointsEdit->value());
-        double step = (points - 1.0) / log10(fabs((stop < 1.0 ? 1.0 : stop) / (start < 1.0 ? 1.0 : start)));
+        double step = (points - 1.0) / log10(fabs((stop) / (start)));
         sweepParamWidget["Step"]->setValue(misc::num2str(step));
       }
       else if (property == "Step")
       {
         double step = str2num(sweepParamWidget["Step"]->value());
-        double points = log10(fabs((stop < 1.0 ? 1.0 : stop) / (start < 1.0 ? 1.0 : start))) * step + 1.0;
+        double points = log10(fabs((stop) / (start))) * step + 1.0;
         pointsEdit->setValue(QString::number(round(points), 'g', 16));
       }     
     }
~~~

@iwbnwif What was the purpose of `stop < 1.0 ? 1.0 : stop` in the equation? The `start` and stop value may be less than one. The value greater than zero must be allowed.

ra3xdh, 2025-01-08:
Log limits calculation is fixed by #1198. The disable of log+DC sweep will be implemented as part of #974, because it depends on simulator.

closed 2025-01-08 by ra3xdh

iwbnwif, 2025-01-17:
> @iwbnwif What was the purpose of stop < 1.0 ? 1.0 : stop in the equation? The start and stop value may be less than one. The value greater than zero must be allowed.

Firstly my apologies for not responding sooner due to travel.

I believe the guards were there to prevent two things:
1. Prevent divide by 0 if the user enters 0 for start
2. Prevent a pole error if the user enters 0 for stop.

However, I think this is no longer necessary since #1079 as they are now floating point numbers which in my understanding resolve to nan / inf in modern compilers? This seems to be the case in the Windows binary but I haven't tried the Linux one yet.

For reference, I was attempting to recreate the previous logic so I can now see should be <= 0.0 rather than < 1.0:

`
misc::str2num(editStop->text(), x, Unit, Factor);
x *= Factor;
misc::str2num(editStart->text(), y, Unit, Factor);
y *= Factor;

if(y == 0.0)  y = x / 10.0;
if(x == 0.0)  x = y * 10.0;
**if(y == 0.0) { y = 1.0;  x = 10.0; }** 
x = (editNumber->text().toDouble() - 1) / log10(fabs(x / y));
`

PR #1198 (2025-01-08): Fix log calculation in componentdialog.cpp
This PR fixes #1195. It reverts old behavior in log limit calculation.