ra3xdh/qucs_s #1205: Cant' run dc simulation with Xyce

adrienluitot, 2025-01-15:
Hello,

I'm using Qucs-S 24.4.1 and Xyce 7.8.
 I try to run dc simulation with Xyce, but it doesn't work, but ac does. Here is my schematic:

[image]

1. If I use Qucs-S' "dc simulation" it generates an empty simulation, nothing is done nor displayed in the log.
2. If I use the Xyce Script for dc, the dc simulation is done but I can't plot anything.
3. If I use the netlist generated by Qucs-S and simulate with Xyce in my terminal, I can see the output results.
4. if I use the Xyce Script for ac or  Qucs-S' "ac simulation", it works and I can plot things.

I don't know if I do things wrong or if there's a bug with Qucs-S?

ra3xdh, 2025-01-15:
It is not a bug. DC is OP simulation. It computes operating point and applied automatically in XYCE mode. The standalone DC simulation has no effect. You need to attach parameter sweep to DC simulation. Read this manual, section 5.2 and 4.2 https://ra3xdh.github.io/pdf/qucs_s_tutorial.pdf The manual is referencing Ngspice. The procedure for Xyce is the same.

ra3xdh, 2025-01-15:
Also insert the current probe in schematic to see the drain current.

closed 2025-01-15 by ra3xdh

adrienluitot, 2025-01-15:
Thank you, sorry I didn't know about that behaviour, that's a little confusing.

DC simulations works now thank you. However I have another question:
How can I plot the variables from "nutmeg", with ngspice I had to use the `.save `command, but I don't think there is an equivalent with Xyce.

[image]

I can't see the `gm_t` and `test` variables in the diagram.

adrienluitot, 2025-01-16:
> Nutmeg scripts works only for Ngspice. It is Ngspice extension. There no equivalent for Xyce.

Okay, good to know. Thus, is there a way to print/plot Xyce's output variable ssuch as `N(XMN0:X1:YPSP103!M1:gm)` ?