ra3xdh/qucs_s #1207: Export as eps or printing to file fails.

Rmano, 2025-01-17:
Version: release 24.4.1
OS: Manjaro Linux up-to-date (basically, arch with extras...)

I checked it with a locally installed package, and with the flatpack version.

This is the circuit: 
[fly-back-parade.sch.zip](https://github.com/user-attachments/files/18454736/fly-back-parade.sch.zip)

[image]

If I try to export the image in `.eps`, it fails with "Inkscape start error" and then `Disk write error`.  Inkscape is correctly installed, as a flatpak with a shell wrapper:

```bash
   ~/bin  inkscape --version                                                        ✔  6s  
Inkscape 1.4 (e7c3feb100, 2024-10-09)
```
If I try to print to file, the PDF come out garbled: 

[prova.pdf](https://github.com/user-attachments/files/18454806/prova.pdf)

[image]

Exporting to PNG works ok.

ra3xdh, 2025-01-17:
I cannot reproduce this. I didn't test the flatpak package, but the locally installed the latest Qucs-S version from Git works as expected. I am using Inkscate 1.4 too. 

>  Inkscape is correctly installed, as a flatpak with a shell wrapper:

You cannot run an external application from flatpak. It is not allowed even with filesystem RW permission because of isolation concept. This cannot be fixed.

Rmano, 2025-01-17:
Yep, I supposed that flatpack would not work. My locally installed was from some packager for arch, (I remember having downloaded a `.zst` file) but I can't find it now...

I'll try to build it here. 

About Inkscape: the app which is a flatpak is Inkscape, not Qucs-s --- I was supposing that I can call Inkscape from the command line all the same from another app. Am I wrong?

ra3xdh, 2025-01-17:
>  My locally installed was from some packager for arch, 

The Qucs-S is available form AUR for a long time: https://aur.archlinux.org/packages/qucs-s and https://aur.archlinux.org/packages/qucs-s-git No need to build from tarball manually. 

> I was supposing that I can call Inkscape from the command line all the same from another app.

I have no Flatpak in my system, but I suspect it may have some restrictions. The same version of Inkscape from the repository works as expected.

closed 2025-01-17 by ra3xdh

Rmano, 2025-01-17:
Yes, the part for `eps` export is fixed, but the print-to-PDF no. I'll try to dig out where the problem can be, maybe it's my printer configuration...

Thanks again

ra3xdh, 2025-01-18:
>  but the print-to-PDF no

I cannot reproduce this issue too. The print to PDF works on my setup as expected. Qucs-S version from Git, KDE, Arch(Manjaro), Qt6.8.1

Rmano, 2025-01-22:
I reported a but to Inkscape: https://gitlab.com/inkscape/inbox/-/issues/11617