ra3xdh/qucs_s #1209: Parameter sweep result is not as expected

windilt, 2025-01-19:
As show in the screen shot, I try to find the proper parameter with Parameter sweep function.
The gain should increase with increased R2, but when I draw the curve, the result show it is stable.

Maybe I didn't draw it in the right way. I searched a lot and tried a lot, seems it is a bug.

[image]

windilt, 2025-01-19:
This is the warning message for this simulation

Ngspice启动...

Note: No compatibility mode selected!

Circuit: * qucs 24.4.0  /home/fedora/qucsworkspace/t4.sch

Doing analysis at TEMP = 27.000000 and TNOM = 27.000000

Using SPARSE 1.3 as Direct Linear Solver
Warning: Interpolated raw file data!

Initial Transient Solution
--------------------------

Node                                   Voltage
----                                   -------
out                                          0
in                                           0
v1#branch                                    0

No. of Data Rows : 199
binary raw file "spice4qucs.tr1._swp.plot"
Doing analysis at TEMP = 27.000000 and TNOM = 27.000000

Using SPARSE 1.3 as Direct Linear Solver
Warning: Interpolated raw file data!

Initial Transient Solution
--------------------------

Node                                   Voltage
----                                   -------
out                                          0
in                                           0
v1#branch                                    0

No. of Data Rows : 199
binary raw file "spice4qucs.tr1._swp.plot"
Doing analysis at TEMP = 27.000000 and TNOM = 27.000000

Using SPARSE 1.3 as Direct Linear Solver
Warning: Interpolated raw file data!

Initial Transient Solution
--------------------------

Node                                   Voltage
----                                   -------
out                                          0
in                                           0
v1#branch                                    0

No. of Data Rows : 199
binary raw file "spice4qucs.tr1._swp.plot"
Doing analysis at TEMP = 27.000000 and TNOM = 27.000000

Using SPARSE 1.3 as Direct Linear Solver
Warning: Interpolated raw file data!

Initial Transient Solution
--------------------------

Node                                   Voltage
----                                   -------
out                                          0
in                                           0
v1#branch                                    0

No. of Data Rows : 199
binary raw file "spice4qucs.tr1._swp.plot"
Doing analysis at TEMP = 27.000000 and TNOM = 27.000000

Using SPARSE 1.3 as Direct Linear Solver
Warning: Interpolated raw file data!

Initial Transient Solution
--------------------------

Node                                   Voltage
----                                   -------
out                                          0
in                                           0
v1#branch                                    0

No. of Data Rows : 199
binary raw file "spice4qucs.tr1._swp.plot"
Reset re-loads circuit * qucs 24.4.0  /home/fedora/qucsworkspace/t4.sch

Circuit: * qucs 24.4.0  /home/fedora/qucsworkspace/t4.sch

ngspice-44 done

ra3xdh, 2025-01-19:
> This is the warning message for this simulation
> Note: No compatibility mode selected!

It is not a warning. It only indicates that the LTSpice/HSPICE compatibility mode is not set and Ngspice works in default mode. This message may be ignored.

> The gain should increase with increased R2, but when I draw the curve, the result show it is stable.

You are using equations wrong. I noticed the following mistakes:

* You cannot derive gain from transient simulation. You should use AC instead.
* You are using Qucsator equaiton in SPICE mode. It serves only for backward compatibility when SPICE mode is active. Normally you even cannot insert in the schematic when SPICE mode is active. How did you managed to insert this equation? You should use Nutmeg equation and attach it to the necessary simulation. 

Please read this tutorial chapter 4: https://ra3xdh.github.io/pdf/qucs_s_tutorial.pdf It explains how to derive AC gain of the RC circuit. Also refer to this list of the documentation: https://github.com/ra3xdh/qucs_s/discussions/788 One of the articles in the list explains the equations usage. Unfortuntelly it is in Russain only, but the automatic translation works fine. 

The documentation update is in progress #554 Don't use manuals from Qucs (without S). It is not compatible anymore. The manual fro readthedocs is 100% outdated. It should not be used too.

closed 2025-01-19 by ra3xdh