ra3xdh/qucs_s #1214: Digital plotting issue with .OPTION INTERP

tomhajjar, 2025-01-20:
Something in Qucs-S has changed over the past few weeks. An PLL simulation stopped generating plots when using Qucs-S but not when using DuSpice. 

PLL-XSPICE.sch is the schematic. Plots don't work anymore if INTERP is used yet PLL-XSPICE_R.sch which is almost identical does generate plots. The choices in the Plot Vs. is full of  *_steps choices.

The following netlists made from PLL-XSPICE.sch both work using DuSpice.
PLL-XSPICE.cir
PLL-XSPICE_interp.cir

[image]
[image]
[image]

[PLL_XSPICE_prj.zip](https://github.com/user-attachments/files/18482321/PLL_XSPICE_prj.zip)

ra3xdh, 2025-01-21:
INTERP is not required anymore. It was needed as a workaround for digital nodes plotting. Otherwise it may break the separate X-axis variable that is used by digital node. Don't use this with digital nodes.

closed 2025-01-21 by ra3xdh

tomhajjar, 2025-01-21:
Will Qucs-S ever be as fast as DuSpice?

Removing INTERP makes some simulations take even longer to process the data file produced by ngspice.

ra3xdh, 2025-01-21:
> Will Qucs-S ever be as fast as DuSpice?

This is beyound the scope of the question. The GUI frontend is always slower that CLI application. The DuSpice uses direct calls to WinAPI, but Qucs-S uses Qt toolkit. It will be always slower.