ra3xdh/qucs_s #1217: ideal source parametrization

dwarning, 2025-01-23:
Using qucs-s 24.4.1 don't allow complete i source parametrization.

[image]

I am not able to set dc value amongst others.

But netlist is OK:

`
R1 0 _net0  1K tc1=0.0 tc2=0.0

I1 0 _net0  DC 0 SIN(0 1M 1G 0 0 0) AC 1M ACPHASE 0
`
Same is happen for ideal v source. 

Using the line configurable I/V source (red) helped me.

ra3xdh, 2025-01-23:
The AC voltage source has the `VO` parameter. It is offset voltage (spice only). The AC current source doesn't have this extension.

dwarning, 2025-01-23:
But why the netlist differs complete? I would expect to parametrize all instance parameters under the GUI.

ra3xdh, 2025-01-23:
The source were initially designed for Qucsator. It has no offset parameter for AC sources. I forgot to add SPICE VO parameter to AC current source.

ra3xdh, 2025-01-23:
The parameters for AC current source added by #1218

dwarning, 2025-01-23:
In spice context VO and IO are the offset values for sinus (transient) waveform description.
Would it not better to speak about ac and dc components which are independent from transient description.
Let us configure all the parameters you are writing in netlist!

ra3xdh, 2025-01-23:
> Let us configure all the parameters you are writing in netlist!

It's too late to change the sources design, because the backward compatibility will be broken and all exiting schematics will be broken. I didn't design the both sources. The sources were initially designed for Qucsator which has no different setup for AC and DC levels. Use the generic source (red one) if you need a different configuration.

dwarning, 2025-01-23:
OK, understand.

PR #1218 (2025-01-23): Add offset and Td parameters for AC current source
This PR fixes #1217. Added IO and TD parameters for AC current source. The both parameters are SPICE-only.

[image]
closed 2025-01-23 by dwarning