ra3xdh/qucs_s #1223: Subcircuit instance model name are unresolved in CDL netlisting.

ThomasZecha, 2025-01-29:
This is a follow-up of the discussion from https://github.com/IHP-GmbH/IHP-Open-PDK/issues/336.

Summary:
In a CDL netlist device hierarchy all nodes must be subcircuits, all leafs must be primitive devices.
This rule is violated in the present case.

One possible solution would be (from original):

_Introduce the following per global settings on/off switchable special treatment of subcircuit instances during CDL netlisting:_

_As a preparation step during CDL netlist export:_

1. _Found all subcircuit instances in the current schematic._
2. _For every instance from 1. check whether the referenced subcircuit exists in the current schematic._
3. _For every instance from 2. where the subcircuit don’t exist do:_
a. _Check whether the second letter of the instance name refer a CDL primitive device (R, C, L , M, Q, ...)_
b. _If a. holds check whether the number of pins match the allowed pin-numbers of the primitive device from a._
c. _If b. holds remove the first letter from the instance name for the CDL-netlisted instance name._

ra3xdh, 2025-01-29:
The solution may be to introduce a property named `CDL_letter` for all devices in *Microelectonics group*. The default is the same as SPICE letter. Current inplementation of `cdl_netlist()` method returns the `spice_netlist`. A separate handling may be required.

ThomasZecha, 2025-02-04:
> The solution may be to introduce a property named CDL_letter for all devices in Microelectonics group. The default is the same as SPICE letter. Current inplementation of cdl_netlist() method returns the spice_netlist. A separate handling may be required.

Since the affected device is located in a library I don't think this is a viable option.

ThomasZecha, 2025-02-19:
> Introduce the following per global settings on/off switchable special treatment of subcircuit instances during CDL netlisting:
> 
> As a preparation step during CDL netlist export:
> 
> Found all subcircuit instances in the current schematic.
> For every instance from 1. check whether the referenced subcircuit exists in the current schematic.
> For every instance from 2. where the subcircuit don’t exist do:
> a. Check whether the second letter of the instance name refer a CDL primitive device (R, C, L , M, Q, ...)
> b. If a. holds check whether the number of pins match the allowed pin-numbers of the primitive device from a.
> c. If b. holds remove the first letter from the instance name for the CDL-netlisted instance name.

@ra3xdh What do you think?

Ikram-rs22, 2025-02-19:
Is there any way to check lvs of qucs-s netlist in klayout then??

ThomasZecha, 2025-02-19:
Yes, see klayout lvs rundecks under https://github.com/IHP-GmbH/IHP-Open-PDK/tree/dev/ihp-sg13g2/libs.tech/klayout/tech/lvs

ra3xdh, 2025-02-19:
> What do you think?

This may resolve the issue. Try to concentrate the proposed changes in the `CdlNetlistWriter.cpp` and `Component::cdl_netlist` The SPICE netlists must not be affected.

Ikram-rs22, 2025-02-25:
Are s2p files accurately netlisted? S2P file of a tx line shows a lot of frequencies in the netlist and klayout lvs check also stops as it cannot properly read the file. Does the tx line in layout have to have a netname for it to be recognized?

ra3xdh, 2025-02-25:
Ngspice doesn't support s2p files directly. It is converted to the table source. Probably CDL should use another entry.

ThomasZecha, 2025-02-25:
Transmission lines are not supported by CDL.

ThomasZecha, 2025-03-06:
I'm currently bussy, but this item is already in my working queue.

closed 2025-05-06 by ra3xdh