ra3xdh/qucs_s #1224: How may I plot  S parameter curves in dB unit ?

wsteffe, 2025-02-01:
Let consider the annexed test case.

[monoliticOnLTCC_qucs.zip](https://github.com/user-attachments/files/18627895/monoliticOnLTCC_qucs.zip)

As you may see the S parameters are not plotted in dB units as usual for this kind of data.

I have tried to put dB(S11) instead of S11 into the trace expression but it does'nt work.
Any hint ?

Thanks

ra3xdh, 2025-02-01:
I cannot reproduce the plotting issue. The attached schematic doesn't contain the equations. I guess you didn't use the equations or using the equations wrong. Refer to the examples in the `share/qucs-s/examples/qucsator` subdirectory. It contains an example of dB plotting with S-parameters. Keep in mind that Nutmeg equation will not work in SPICE mode.  

Also the archive doesn't contain the s2p file. The schematic will not work without it.  

[image]

andresmmera, 2025-02-01:
@wsteffe You can try with the example attached. Simply replace the S2P file as needed.

As @ra3xdh said, the equation block is missing.

[Plot_S2P_prj.zip](https://github.com/user-attachments/files/18628206/Plot_S2P_prj.zip)

wsteffe, 2025-02-01:
Hello ra3xdh. Many thanks for help.
In fact using qucsator and variables dBS11, dBS12 defined in equations it works.

I had already tried a similar approach before (using both nutmag or qucsator equations) but it gave me an error (indefinite variable or something similar) I do not know what has changed but now it is fixed.

Many tanks

ra3xdh, 2025-02-01:
> I had already tried a similar approach before (using both nutmag or qucsator equations)

I guess you have used qucsator syntax with SPICE or SPICE syntax with Qucsator. Different simulation kernels require different S-parameter equation syntax. Qucsator equations supports only limited syntax subset in SPICE mode. Refer to the examples shipped with Qucs-S distribution for S-parameters simulation examples with equations. It contains examples for both Ngspice and Qucsator. 

Closing this issue as resolved and transferring to Discussions/Q&A