ra3xdh/qucs_s #1233: .GLOBAL Variables

tomhajjar, 2025-02-08:
In reference to #1227, I tried to make a Data Bus sub-circuit but cannot figure out create the following statement in the subcircuit:

.GLOBOL D0 D1 D2 D3 D4 D5 D6 D7

[image]

[image]

[image]

[Digital_Data_Bus_prj.zip](https://github.com/user-attachments/files/18716317/Digital_Data_Bus_prj.zip)

ra3xdh, 2025-02-08:
Please suspend the work on buses implementation. This should be resolved in C++ level. Adding bus as subcircuit device is not a good solution.

ra3xdh, 2025-02-08:
I have created a task for implementing buses in schematic editor. Maybe someone will implement it. See #1234

closed 2025-02-08 by ra3xdh