ra3xdh/qucs_s #1256: Qucs Spice not detecting ngspice simulator

jgardona, 2025-02-24:
Hello, good evening.

Notice that in new Qucs Spice 25.1.0, the ngspice_con.exe is not being detected. We cant simulate.

[image]

so - win 11 x86_64
qucs-s version - 25.1.0

jgardona, 2025-02-24:
Actually, I needed to download ngspice from its web page. The default one is not working. Mabe wrong architecture?

ra3xdh, 2025-02-24:
I cannot reproduce this problem with Windows 10. I made a fresh installation using the latest installer and everything works as expected. Ngspice is detected. Does application find QucsatorRF? I can only suspect this may be related to UAC settings in Windows 11 preventing access to installation directory. 

[image]

jgardona, 2025-02-24:
> I cannot reproduce this problem with Windows 10. I made a fresh installation using the latest installer and everything works as expected. Ngspice is detected. Does application find QucsatorRF? I can [...]

Hello, good morning.

I cant reproduce anymore with a fresh installation also. Will close this issue. Maybe was an older installation in appData folder.

closed 2025-02-24 by jgardona