ra3xdh/qucs_s #1257: Simplify FFT analysis with ngspice

dwarning, 2025-02-24:
In case I want make transient and FFT analysis on same schematic I have to conduct two transient analysis's. This is cumbersome and tedious. I am missing the same approach as we can use in ngspice script by making one transient, plot the signals and then doing fft on same data. 
The solution for this feature would be to enhance the FFT widget with same Simulation entry as we have for parametric sweep. So the user can choose a previous transient simulation or left it out. In the latter case the transient simulation will be activated as we have actual.

tomhajjar, 2025-02-26:
I never had to do two trans

[AC_Tran_FFT_prj.zip](https://github.com/user-attachments/files/18988107/AC_Tran_FFT_prj.zip)

dwarning, 2025-02-26:
Thanks for spotting that.

But you write in your files:

`
Qucs-S sends 2 transient simulations to ngspice

1st used for V/I versus time
2nd used for S versus frequency
`
At the moment the simulations crashing the simulator. Must see, maybe I have to set compatibility mode?

ra3xdh, 2025-02-27:
The redesign of the FFT analysis is not possible at the current time point because the backward compatibility with existing schematics will be lost. Old schematics will not load if the simulation will be modified. I cannot implement the proposed modifications.

closed 2025-02-27 by ra3xdh

dwarning, 2025-02-27:
Why not implement an additional simulation widget which has this capability? Will not affect older schematics.