ra3xdh/qucs_s #1260: Options: Unable to add/modify .OPTIONS parameters

tskaar, 2025-02-27:
__Description__
Currently there's no way to modify the "Name" field of the .OPTIONS blocks, causing the defaults (XyceOptionsPackage & GMIN) being the only parameters available.

Previously (v24.4.1) one was able to add/modify multiple parameters in the .OPTIONS block.

v25.1.0:
[image]

v24.4.1:
[image]
Temp added just as an example of "typical" use case for the .OPTIONS block

__Expectation__
Being able to Add/Modify the parameters per .OPTION block

__Information__
Version: v25.1.0
OS: Archlinux
Package: [AUR](https://aur.archlinux.org/packages/qucs-s)

ra3xdh, 2025-02-27:
The most probably this was a result of #1054. We missed this point while testing. @iwbnwif Could you have a look?

ra3xdh, 2025-02-27:
@tskaar You may use INCLUDE SCRIPT device as the workaround. It allows to specify the .OPTION directive. 

[image]

tskaar, 2025-02-27:
@ra3xdh;
That's a good tip/workaround, thanks!

iwbnwif, 2025-02-28:
I think that the practical way of solving this is to treat the .OPTIONS component in the same way as we do equations. I will look at this later today.

ra3xdh, 2025-02-28:
Yes, the both .OPTIONS and .FUNC need to switch to free editing mode.

closed 2025-04-07 by ra3xdh