ra3xdh/qucs_s #1273: Crash in Schematic::oneTwoWires when dragging overlapping wires

SmallJoker, 2025-03-20:
The following schematic crashes Qucs-S on Windows and Linux when trying to drag `D1`: [double_wire_drag_crash.sch.txt](https://github.com/user-attachments/files/19371446/double_wire_drag_crash.sch.txt)

Removing the wires one by one reveals a wire segment located under another wire. This is likely the source of this crash.

Reproduced on:
* Linux AppImage, version 24.4.99 (2024-11-03)
* Windows, version 25.1.1

Linux backtrace: (Interestingly this freezes most of my X11 desktop if the program does not terminate immediately)
```
$ gdb -q --batch -iex "set confirm off" -ex run -ex bt --args Qucs-S-24.4.99-linux-x86_64.AppImage

<snip>

Thread 1 "AppRun.wrapped" received signal SIGSEGV, Segmentation fault.
0x00005555556d035b in std::__cxx11::list<Element*, std::allocator<Element*> >::remove[abi:__cxx20](Element* const&) [clone .isra.0] ()
#0  0x00005555556d035b in std::__cxx11::list<Element*, std::allocator<Element*> >::remove[abi:__cxx20](Element* const&) [clone .isra.0] ()
#1  0x00005555556d1dd3 in Schematic::oneTwoWires(Node*) ()
#2  0x00005555556d92cc in Schematic::copySelectedElements(Q3PtrList<Element>*) ()
#3  0x00005555556dfd23 in MouseActions::MMoveMoving(Schematic*, QMouseEvent*) ()
#4  0x00005555556961ed in Schematic::contentsMouseMoveEvent(QMouseEvent*) ()
#5  0x0000555555c45ed4 in Q3ScrollView::viewportMouseMoveEvent(QMouseEvent*) ()
#6  0x0000555555c454f8 in Q3ScrollView::eventFilter(QObject*, QEvent*) ()
#7  0x00007ffff67eaefa in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) () from /tmp/.mount_Qucs-ScsSFOe/usr/bin/../lib/libQt6Core.so.6
#8  0x00007ffff78ed4e1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /tmp/.mount_Qucs-ScsSFOe/usr/bin/../lib/libQt6Widgets.so.6
#9  0x00007ffff78f5e10 in QApplication::notify(QObject*, QEvent*) () from /tmp/.mount_Qucs-ScsSFOe/usr/bin/../lib/libQt6Widgets.so.6
#10 0x00007ffff67eb18a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /tmp/.mount_Qucs-ScsSFOe/usr/bin/../lib/libQt6Core.so.6
#11 0x00007ffff78f5160 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) () from /tmp/.mount_Qucs-ScsSFOe/usr/bin/../lib/libQt6Widgets.so.6
#12 0x00007ffff7956a81 in ?? () from /tmp/.mount_Qucs-ScsSFOe/usr/bin/../lib/libQt6Widgets.so.6
#13 0x00007ffff79596eb in ?? () from /tmp/.mount_Qucs-ScsSFOe/usr/bin/../lib/libQt6Widgets.so.6
#14 0x00007ffff78ed4f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /tmp/.mount_Qucs-ScsSFOe/usr/bin/../lib/libQt6Widgets.so.6
#15 0x00007ffff67eb18a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /tmp/.mount_Qucs-ScsSFOe/usr/bin/../lib/libQt6Core.so.6
#16 0x00007ffff6f33600 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () from /tmp/.mount_Qucs-ScsSFOe/usr/bin/../lib/libQt6Gui.so.6
#17 0x00007ffff6f8d77c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /tmp/.mount_Qucs-ScsSFOe/usr/bin/../lib/libQt6Gui.so.6
#18 0x00007ffff2b0fcda in ?? () from /tmp/.mount_Qucs-ScsSFOe/usr/plugins/platforms/../../lib/libQt6XcbQpa.so.6
#19 0x00007ffff5d26d3b in g_main_context_dispatch () from /tmp/.mount_Qucs-ScsSFOe/usr/bin/../lib/libglib-2.0.so.0
#20 0x00007ffff5d7c2b8 in ?? () from /tmp/.mount_Qucs-ScsSFOe/usr/bin/../lib/libglib-2.0.so.0
#21 0x00007ffff5d243e3 in g_main_context_iteration () from /tmp/.mount_Qucs-ScsSFOe/usr/bin/../lib/libglib-2.0.so.0
#22 0x00007ffff6aa276a in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /tmp/.mount_Qucs-ScsSFOe/usr/bin/../lib/libQt6Core.so.6
#23 0x00007ffff67f6963 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /tmp/.mount_Qucs-ScsSFOe/usr/bin/../lib/libQt6Core.so.6
#24 0x00007ffff67f341e in QCoreApplication::exec() () from /tmp/.mount_Qucs-ScsSFOe/usr/bin/../lib/libQt6Core.so.6
#25 0x0000555555679ca9 in main ()
``` 

Windows backtrace: (also `gdb`)
```
Thread 1 received signal SIGSEGV, Segmentation fault.
0x00007ff7819e61cc in ?? ()
(gdb) bt
#0  0x00007ff7819e61cc in ?? ()
#1  0x00007ff7819e7d88 in ?? ()
#2  0x00007ff7819ef78c in ?? ()
#3  0x00007ff7819f9c34 in ?? ()
#4  0x00007ff7819ade52 in ?? ()
#5  0x00007ff781ff95d8 in ?? ()
#6  0x00007ff781ff46e7 in ?? ()
#7  0x00007ffe83fa8f20 in ?? ()
   from XXXX\qucs_s_win64\bin\Qt6Core.dll
#8  0x00007ffe826863b5 in ?? ()
   from XXXX\qucs_s_win64\bin\Qt6Widgets.dll
#9  0x00007ffe8269105e in ?? ()
   from XXXX\qucs_s_win64\bin\Qt6Widgets.dll
#10 0x00007ffe83fa9a20 in ?? ()
   from XXXX\qucs_s_win64\bin\Qt6Core.dll
#11 0x00007ffe8268f6bf in ?? ()
   from XXXX\qucs_s_win64\bin\Qt6Widgets.dll
#12 0x00007ffe826e9701 in ?? ()
   from XXXX\qucs_s_win64\bin\Qt6Widgets.dll
#13 0x00007ffe826ec2cb in ?? ()
   from XXXX\qucs_s_win64\bin\Qt6Widgets.dll
#14 0x00007ffe826863c5 in ?? ()
   from XXXX\qucs_s_win64\bin\Qt6Widgets.dll
#15 0x00007ffe83fa9a20 in ?? ()
   from XXXX\qucs_s_win64\bin\Qt6Core.dll
#16 0x00007ffe83648f26 in ?? ()
   from XXXX\qucs_s_win64\bin\Qt6Gui.dll
#17 0x00007ffe8369dce3 in ?? ()
   from XXXX\qucs_s_win64\bin\Qt6Gui.dll
#18 0x00007ffe8416e5c6 in ?? ()
   from XXXX\qucs_s_win64\bin\Qt6Core.dll
#19 0x00007ffe839f8889 in ?? ()
   from XXXX\qucs_s_win64\bin\Qt6Gui.dll
#20 0x00007ffe83fb55d4 in ?? ()
   from XXXX\qucs_s_win64\bin\Qt6Core.dll
#21 0x00007ffe83fb3575 in ?? ()
   from XXXX\qucs_s_win64\bin\Qt6Core.dll
#22 0x00007ff7819a6bde in ?? ()
#23 0x00007ff782002a4c in ?? ()
#24 0x00007ff781991319 in ?? ()
#25 0x00007ff781991406 in ?? ()
#26 0x00007fff4c84dbe7 in KERNEL32!BaseThreadInitThunk ()
   from C:\WINDOWS\System32\kernel32.dll
#27 0x00007fff4cd45a6c in ntdll!RtlUserThreadStart ()
   from C:\WINDOWS\SYSTEM32\ntdll.dll
#28 0x0000000000000000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
```

ra3xdh, 2025-03-20:
I confirm this bug. This may be related to the wire planner. @wawuwo , please have a look.

wawuwo, 2025-03-21:
@SmallJoker, can you please provide a more detailed description how to reproduce the issue? I have no problem dragging `D1` using build from the head of the `current` branch:

https://github.com/user-attachments/assets/7ee8d53b-f25a-404d-98b5-3e068cacf7e8

Though it's strange that the wire just beneath the D1 consists of two segments… They are supposed to be merged by the schematic editor. Did you edited the schematic by hand i.e. as a text? That's rather strange 🤔

wawuwo, 2025-03-21:
There is definitely something wrong with that wire, but I don't understand how it got like this in the first place

https://github.com/user-attachments/assets/cba02fad-ad54-43af-b07b-e3a80a83f195

SmallJoker, 2025-03-21:
@wawuwo I think I managed to get the wires like this by copying and moving parts of the circuit. Unfortunately I don't have any reproduction steps. Regardless, Qucs-S should not crash. Here's how I did it: (tried to move D1 slightly upwards)

https://github.com/user-attachments/assets/e78ca66a-c215-411a-965c-0483199d6c32

wawuwo, 2025-03-25:
Well I now why those wire segments are not merged — there is a node between them, I had to install the libs.

The problem was in this line in schematic file:
```
  <510 350 350 350 "" 0 0 0 "">
```
— x1 is larger than x2. So the easiest way to not crash the app when opening this file is to change it manually to
```
  <350 350 510 350 "" 0 0 0 "">
```

I've added a "normalisation" step before inserting wire to schematic, so with the fix merged such wires shouldn't be a problem anymore.

https://github.com/user-attachments/assets/e08cd0fa-975f-4d26-9422-48f846b92927

I'm also attaching two schematics with "bad" wires as an example if someone is interested in comparing behaviours.

[hor-ver-crash.zip](https://github.com/user-attachments/files/19455400/hor-ver-crash.zip)

closed 2025-03-25 by wawuwo

SmallJoker, 2025-03-25:
That's great. Thank you for the fix!

commit 565efd3b (2025-03-25): Merge pull request #1278 from wawuwo/fix-1273-crash-in-one-two-wires

Fix crash when coordinates of wire ends are in wrong order

Fix #1273