ra3xdh/qucs_s #1282: Logarhitmic graphing completely broken

Maksymisio, 2025-04-03:
OS: Windows 10 Version: 25.1 
Selecting the logarhitmic axis options breaks the graphs, completely showing a blank graph
this has been an issue for a very long time
[image]

Maksymisio, 2025-04-03:
ok sorry it's not completely broken but inconsistent, you have to manually specify the range multiple times to make it work

ra3xdh, 2025-04-04:
It is not a bug. Logarithmic units cannot be computed from the negative values and/or zero. Transient data often contains negative values. Maybe it would be a good solution to add a more clear warning message for user instead of "invalid data". Something like "cannot compute log() from negative data".

closed 2025-04-04 by ra3xdh