ra3xdh/qucs_s #1284: Magnetic cores

ra3xdh, 2025-04-05:
Qucs-S provides two devices related to magnetic cores simulation in the system library. Both could be found in the group *XSPICE devices*:
* XSPICE Icouple
* XSPICE core

The both devices seems to be broken at the simulator level. See this thread at the Ngspice forum: https://sourceforge.net/p/ngspice/discussion/120973/thread/b36bb618/ The simulation works, but gives a wrong result. For example I failed to create a model for Amidon 43 ferrite. The attached example from the Ngspice documentation uses a strange units for BH curve. It is not possible to create a practical magnetic core model using the existing XSPICE devices. These devices are only confusion for a user. 

Currently I am working on a replacement library using the Jiles-Atherton model https://en.wikipedia.org/wiki/Jiles%E2%80%93Atherton_model for saturated core with hysteresis. This model is more advanced than PWL one used by XSPICE. My proposal is following:

* Provide a replacement library containing a models for commonly used ferrite and iron cores like N87, Amidon, etc.
* Deprecate the XSPICE Icouple and Core devices. Move the both to library and exclude from the left panel.

tomhajjar, 2025-04-08:
It's unclear if anyone uses Transformer.lib but it does have XSPICE Icouple and Core devices. Is there a good reason to eliminate them? Where do people find parameters for the Jiles-Atherton model?

ra3xdh, 2025-04-08:
> Where do people find parameters for the Jiles-Atherton model?

It may be converted from MicroCAP libraries. The both MicroCAP and PSPICE uses this approach and have a big set of ferrite cores models.

ra3xdh, 2025-04-15:
I have an intermediate result with J.-A. magnetic core model. Here is a simulation example. It uses modelling approach from  https://ieeexplore.ieee.org/document/8267734/authors#authors The result matches MicroCAP simulation using the same modelling technique. But it inherits some limitations of MIcroCAP model like wide hysteresis loops at small signal. See https://dmitry-shem.livejournal.com/28220.html?utm_source=3userpost (in Russian, could be read with automatic translator). 

The logic of the new library is following:

* The core model allows to define A,K,C,MS parameters, core area, and magnetic path length. It could be converted from MicroCAP library. It has two terminals (input H and output B).
* The winding must be connected to H and B terminal of the core. The turns number and series resistance could be defined. 

[image]

[image]

ra3xdh, 2025-04-23:
The most of requested features implemented by #1317. This issue may be closed. I may provide an extension of Magnetic cores support later. It is not a final version.

closed 2025-04-23 by ra3xdh