ra3xdh/qucs_s #1308: Wire: Double-clicking wire to end wire-drawing crashes application with "Assertion `first != last' failed"

tskaar, 2025-04-16:
Hi!

Typically when I draw wires I end the wire-drawing by double clicking the same point twice, however that crashes the application with the following assertion at the moment:

```bash
qucs/geometry/multi_point.h:212: std::vector<P> qucs_s::geom::on_line(P, P, It, It) [with P = Node*; It = std::_List_iterator<Node*>]: Assertion `first != last' failed.
```

I'm not a C++ developer, however the issue can be worked-around with the following patch:

```patch
diff --git a/qucs/schematic_element.cpp b/qucs/schematic_element.cpp
index bbeec922..bddb7a35 100644
--- a/qucs/schematic_element.cpp
+++ b/qucs/schematic_element.cpp
@@ -2264,6 +2264,10 @@ std::pair<bool,Node*> Schematic::connectWithWire(const QPoint& a, const QPoint&
     for (std::size_t i = 1; i < points.size(); i++) {
         auto m = points[i-1];
         auto n = points[i];
+        // Skip overlapping points
+        if (m == n) {
+            continue;
+        }
         auto [ch, node] = installWire(new Wire(m.x(), m.y(), n.x(), n.y()));
         hasChanges = hasChanges || ch;
     }
```

**Test-case**
1. Start drawing wire
2. Double click at the same point to "end" wire drawing
3. Assertion triggers

ra3xdh, 2025-04-16:
This may be related to #1266. @wawuwo Please have a look.

wawuwo, 2025-04-16:
Hey, thanks for the patch! I think it's better to handle it somewhere up the call stack… Double click in this case acts not as "double click" but as series of clicks i.e. it effectively tries to insert a wire of length zero. I think it's better not to even try to do this in the first place. Will take a look 🤝

ra3xdh, 2025-04-17:
Fixed by #1311

closed 2025-04-17 by ra3xdh

tskaar, 2025-04-17:
Confirmed working on my end, thanks for the quick fix guys! 🙌