ra3xdh/qucs_s #1312: CI fails to download Qt

ra3xdh, 2025-04-17:
The latest CI jobs are always failing at the download Qt stage. @dsm, please have a look. Here is the log:

~~~
INFO    : aqtinstall(aqt) v3.1.21 on Python 3.11.11 [CPython GCC 11.4.0]
/opt/hostedtoolcache/Python/3.11.11/x64/bin/python3 -m aqt install-qt linux desktop 6.9.0 linux_gcc_64 --autodesktop --outputdir /home/runner/work/qucs_s/Qt --modules qtcharts
INFO    : aqtinstall(aqt) v3.1.21 on Python 3.11.11 [CPython GCC 11.4.0]
WARNING : Connection to 'https://download.qt.io/' failed. Retrying with fallback 'https://qtproject.mirror.liquidtelecom.com/'.
WARNING : Failed to download checksum for the file 'online/qtsdkrepository/linux_x64/desktop/qt6_690/qt6_690/Updates.xml'. This may happen on unofficial mirrors.
INFO    : Downloading qtdeclarative...
INFO    : Downloading qttranslations...
INFO    : Finished installation of qttranslations-Linux-RHEL_8_10-GCC-Linux-RHEL_8_10-X86_64.7z in 6.37221120
INFO    : Finished installation of qtdeclarative-Linux-RHEL_8_10-GCC-Linux-RHEL_8_10-X86_64.7z in 145.89811115
INFO    : Downloading qtwayland...
INFO    : Downloading icu...
INFO    : Finished installation of qtwayland-Linux-RHEL_8_10-GCC-Linux-RHEL_8_10-X86_64.7z in 53.27574714
INFO    : Downloading qttools...
INFO    : Finished installation of icu-linux-Rhel8.6-x86_64.7z in 40.59998128
INFO    : Finished installation of qttools-Linux-RHEL_8_10-GCC-Linux-RHEL_8_10-X86_64.7z in 75.05666074
WARNING : Caught ChecksumDownloadFailure, terminating installer workers
ERROR   : Failed to download checksum for the file '6.9.0-0-202503301022qtbase-Linux-RHEL_8_10-GCC-Linux-RHEL_8_10-X86_64.7z' from mirrors '['https://download.qt.io/']
==============================Suggested follow-up:==============================
* Check your internet connection
* Consider modifying `requests.max_retries_to_retrieve_hash` in settings.ini
* Consider modifying `mirrors.trusted_mirrors` in settings.ini (see https://aqtinstall.readthedocs.io/en/stable/configuration.html#configuration)
usage: aqt [-h] [-c CONFIG]
           {install-qt,install-tool,install-doc,install-example,install-src,list-qt,list-tool,list-doc,list-example,list-src,help,version}
           ...

Another unofficial Qt Installer.
aqt helps you install Qt SDK, tools, examples and others

options:
  -h, --help            show this help message and exit
  -c CONFIG, --config CONFIG
                        Configuration ini file.

subcommands:
  aqt accepts several subcommands:
  install-* subcommands are commands that install components
  list-* subcommands are commands that show available components

  {install-qt,install-tool,install-doc,install-example,install-src,list-qt,list-tool,list-doc,list-example,list-src,help,version}
                        Please refer to each help message by using '--help' with each subcommand
~~~

dsm, 2025-04-17:
@ra3xdh Yes sometimes it happens, sometimes server connection failed then couldn't download but after sometime rerun ci jobs finished succesfully so it didn't failed this time. I'll look how to improve(more robust) this behaviour.

ra3xdh, 2025-04-22:
The issue seems to be self-fixed recently. Closing.

closed 2025-04-22 by ra3xdh