ra3xdh/qucs_s #1316: Wrong dc bias value calculated in Votalge probe.

jgardona, 2025-04-18:
Hello, good afternoon.

Im seeing wrong dc bias calculation in voltage probe while measure voltage drop in a 1n4148 diode.
Value drop is correct in table.

[image]

Wrong value in voltage probe bias calculation.

[image]

dsm, 2025-04-19:
DC bias calculation is correct. `3.3 - 2.5407 = 0.7593` dc bias gives nodes voltage does not difference between nodes.

jgardona, 2025-04-19:
> DC bias calculation is correct. `3.3 - 2.5407 = 0.7593` dc bias gives nodes voltage does not difference between nodes.

How this is correct? The issue is in voltage probe.

[image]

dsm, 2025-04-19:
I think, the labels overlap each other. this node also 3.3v.

jgardona, 2025-04-19:
> I think, the labels overlap each other. this node also 3.3v.

Yep, you are right. Maybe a not rendering in voltage probe terminals can be a better solution.