ra3xdh/qucs_s #1328: Verilog-A support is not working?

MrQbasic, 2025-04-24:
I would like to build a Verilog-A module but can't. I have installed all the dependencies and set up the Locations correctly in the application settings.

Sorry for the error being in German but it just says file not found.

Also it is missing "Program Files/Qucs-S/include/" entirely. The folder does not exist.

I did not build from source. Would that fix it ?

On Linux(Arch) it's even worse. All the dependencies are out of date and I'd need to install everything by hand that's why im on Windows for this.

[image]

Kreijstal, 2025-04-25:
sounds like a dll could be missing, either that or what happens is that when path is longer than 260 characters windows says NOPE NOT FOUND, NEVER.

Kreijstal, 2025-04-25:
can you share the model?

ra3xdh, 2025-04-25:
It is not a bug. ADMS support is deprecated and ADMS itself and its includes are not shipped with Qucs-S anymore because of Perl build dependencies. The instruction from MOS-AK presentation is totally outdated. Switch to Ngspice+OpenVAF. OpenVAF is much more advanced tool than ADMS. The `openvaf-r` package is available in Arch repositories (AUR). The Ngspice+OpenVAF supports old way of defining Verilog-A devices described in MOS-AK presentation. But it's recommended to use a new more flexible way: https://github.com/ra3xdh/qucs_s/issues/411 Documenting this feature is in progress. 

If you need ADMS by some reasons you have to install one of the old versions before 1.1.0 Keep in mind that ADMS and simulator must be built with the same C++ compiler version. Otherwise you will get a crash.

closed 2025-04-25 by ra3xdh

MrQbasic, 2025-04-25:
Ok so to clear things up. Yes I'm missing all the deps which results in the given error. Thou then i would remove the "Build Verilog-A Module" button from the Menu ?

Also my .va file was just a generic test example of an HJ-Fet Chatgpt gave me to test verilog.

ra3xdh, 2025-04-25:
The *Build Verilog-A module* menu point works. It invokes the OpenVAF compiler when Nsgpice mode is active. What are your application settings? Are you using SPICE mode?

MrQbasic, 2025-04-25:
Oh i see. When selecting Ngspice it works. Thanks.