ra3xdh/qucs_s #1335: large gain woth a passive circuit

wsteffe, 2025-04-29:
The annexed circuit, simulated with qucsator, exhibits a positive gain (in dB) even if composed of passive components alone (resistors,capacitors and gyrators).

[testPassive_prj.zip](https://github.com/user-attachments/files/19956767/testPassive_prj.zip)

ra3xdh, 2025-04-29:
The issue transferred here: https://github.com/ra3xdh/qucsator_rf/issues/38 It is not related to GUI. I suspect the problem may be in gyrator implementation. The controlled source is not passive device.

wsteffe, 2025-04-29:
The problem is that gyrator component is broken.
I have replaced it with a subcircuit that implements the gyrator function with a couple of voltage controlled current sources and it works as expected. See following project:

[PICOM_load_prj.zip](https://github.com/user-attachments/files/19958404/PICOM_load_prj.zip)

ra3xdh, 2025-04-29:
> The problem is that gyrator component is broken.

This proves my suspicions. I will try to fix the gyrator on Qucsator side.