ra3xdh/qucs_s #1343: Workdir not used to store ngspice simulation results

ivandi69, 2025-05-04:
Current branch up to commit 1289ad9f0b940282fdd5289b5ccb1186ac38a8cf on Linux, gcc, Qt-6.9.0, Wayland.

Ngspice simulation result files are stored in ~/.cache instead of ~/.cache/qucs-s.

ra3xdh, 2025-05-04:
I don't observe this bug on machine. The simulator cache is located in `~/.cache/qucs-s.`

ivandi69, 2025-05-04:
If I start qucs-s from current branch with no config in **.config/qucs/qucs_s.conf** I have: **S4Q_workdir=/home/ivandi/.cache**

If I first start qucs-s release 25.1.2 I have **S4Q_workdir=/home/ivandi/.cache/qucs-s** and then qucs-s from current branch works fine.

"File → Application settings → Locations"
[image]

wawuwo, 2025-05-04:
I confirm, without existing configuration file `QStandardPaths::writableLocation(QStandardPaths::CacheLocation)` [here](https://github.com/ra3xdh/qucs_s/blob/1289ad9f0b940282fdd5289b5ccb1186ac38a8cf/qucs/settings.cpp#L74) returns path without application-specific part. I've tested it on Mac + Qt 6.9.0 and Debian + Qt 6.4.2 — the same behaviour in both cases.

closed 2025-05-07 by ra3xdh