ra3xdh/qucs_s #1375: Current branch. File -> Exit does not close the application window.

ivandi69, 2025-05-19:
Up to commit 194db581c37c0b53d9fe2fab6b533ce818e60070

Close button on the title bar works fine if used before File->Exit menu.

File -> Exit closes all documents, but not the application window. In this state close button on the title bar does not work any more and I have to kill the application.

Tested on Windows 10 and Linux.

ra3xdh, 2025-05-19:
Git bisect shows that this has appeared after this commit: https://github.com/ra3xdh/qucs_s/commit/9d08c51e9cfa25fa6b3614cf26447b49541c92c2 @andresmmera , Please take a look.

andresmmera, 2025-05-19:
I fixed that, but, hell, I accidentally pushed directly to current.

@ra3xdh can you revoke my permissions (Collaborator) to avoid things like this?

andresmmera, 2025-05-19:
Should I drop the last two commits and open a new PR or it's better to keep them as they are?

ra3xdh, 2025-05-19:
> I fixed that, but, hell, I accidentally pushed directly to current.

Yes, Now everything works as expected

> can you revoke my permissions (Collaborator) to avoid things like this?

It's need to configure branch protection to avoid such situation in the future: https://docs.github.com/en/repositories/configuring-branches-and-merges-in-your-repository/managing-protected-branches/managing-a-branch-protection-rule

> Should I drop the last two commits and open a new PR or it's better to keep them as they are?

The latest commits on `current` may present.

closed 2025-05-19 by ra3xdh

andresmmera, 2025-05-19:
Ok, I'll do so.
Thank you!