ra3xdh/qucs_s #1382: Invisible DC source of 0.536 Volts

shafinhossain, 2025-05-22:
[image]
As  you can see in the image, I have cancelled out the DC bias source at base of the HBT and only used an AC source followed by a DC block cap. But still there is a consistent 0.536 Volts present, even if I set the initial voltage of the DC block cap at 0 Volts.

ra3xdh, 2025-05-22:
It is most likely not Qucs-S issue. It shows exactly what it gets from simulator. There is not HBT model in the Qucs-S libraries. What is the source of the HBT model?  The 0.536V is most likely related to the leakage current of the HBT device. Also initial conditions are not applied for DC OP according Ngspice specification.

shafinhossain, 2025-05-22:
This is an HBT from IHP 130 nm Open Source PDK. Okay I think I might have understood the reason.

ra3xdh, 2025-05-22:
Closing as resolved.

closed 2025-05-22 by ra3xdh