ra3xdh/qucs_s #1385: Marker jumps to start of diagram when added if diagram has a very narrow custom time/x interval

njascgil, 2025-05-29:
As the title says, if I try to add a marker to a diagram which was configured for a very narrow time interval, the marker does not stay where I click but jumps instead to the left.
Additionally, with the same narrow time interval, the diagram appears to show some artifacts on top of the left y axix.

[image]

I am attaching a file/simulation where both issues I am facing should be visible. Just run the simulation with the Qucsator engine and try to add a Marker somewhere.

[marker_bug.zip](https://github.com/user-attachments/files/20500800/marker_bug.zip)

ra3xdh, 2025-05-29:
I confirm this. This may be related to the recent redesign of diagrams and schematic #1266 @wawuwo , Could you check?

The workaround is to set the X-position in the marker properties dialog:

[image]

wawuwo, 2025-05-29:
I’ll take a look at this on the weekend

image.png (view on web)

wawuwo, 2025-05-31:
I have given it a look, it seems that the problem may be somewhere in "diagraming" code and is not related to #1266. The same behaviour can be observed in rather old Qucs-S 24.1. "Diagraming" code is quite messy, I think solution will take some time.

ra3xdh, 2025-06-01:
Fixed by #1391

closed 2025-06-01 by ra3xdh