ra3xdh/qucs_s #1396: Properties menu for wire is not opening on right-click

Radvall, 2025-06-04:
When right-clicking on a wire in the editor, the properties menu fails to open (see attached video).

https://github.com/user-attachments/assets/66435513-a5e2-478d-b5b8-1a5d00333a4b

ra3xdh, 2025-06-04:
I am not sure if the *Properties* entry was ever implemented for wire. Probably this point should be disabled if the object type is wire.

Radvall, 2025-06-08:
This is my mistake, I initially assumed that the net color could be changed via the "Properties" dialog. However, after checking in the original QUCS, I realized that double-clicking on a net simply opens the same "Properties" dialog as right-click → "Properties".

I would then like to reclassify this issue as a feature request. Would it be possible to add a color selection field to the net properties dialog? This would help improve schematic clarity and visual organization, especially in large circuits.

ra3xdh, 2025-06-14:
Let's keep the things as is. Probably we will add a possibility to define wire colors etc. And the wire properties dialog will be required for this feature. But this is a future task. Closing for now.

closed 2025-06-14 by ra3xdh