ra3xdh/qucs_s #1429: Error message for every simulation

tomhajjar, 2025-07-24:
Ngspice started...
Error opening code model "../lib/ngspice/tlines.cm"
: The system cannot find the file specified.

[image]

ra3xdh, 2025-07-24:
This error message not related to Qucs-S GUI. It comes from Ngspice. I cannot reproduce this error. Both Windows and Linux versions work s expected on my machine. It seems your installation of Ngspice is wrong or incomplete. You may also have $PATH issues and Ngspice looks for libraries in a wrong location.

ra3xdh, 2025-07-24:
Since the version 45 Ngspice will include a new XSPICE library containing transmission lines models. See #1401. It seems this library is missing in your version. This may be because of wrong Ngspice build. Remove the line containing this library from spiceinit if it is nor needed.