ra3xdh/qucs_s #1443: Render issue with Wayland?

VIPQualityPost, 2025-08-16:
I'm not sure if you have many Wayland users but figured I would leave here anyway.
Some components like Inductor have issues creating wires when rotated (maybe render issue? I do not get this behavior on Mac).
When trying to rearrange the wire the application will hard crash.
```
qucs-s: /home/<user>/Desktop/qucs_s/qucs/schematic_element.cpp:2643: void Schematic::dumbConnectWithWire(const QPoint&, const QPoint&): Assertion `a != b' failed.
Aborted (core dumped)

```
This is a clean build off of current branch.

https://github.com/user-attachments/assets/cb787a9c-bf00-4840-9fba-1c8677d14523

ra3xdh, 2025-08-16:
I am able to reproduce this bug with X11. It is not related to Wayland. The assertions have effect only for debug build. This may be related to the recent redesign of schematic editor. @wawuwo What was the purpose of this assertion?

ra3xdh, 2025-08-16:
The assertion takes place if the wire with zero length is formed after the end dragging.

ra3xdh, 2025-08-21:
Fixed by #1449.

PR #1449 (2025-08-21): Change schematic healing logic: extend/shrink wires when possible
Hi!

The changes are in the title, this also should fix #1443. 

* geometry/multi_point.h Add 'is_it_line' function to check if three points form a line

* healer.cpp Adjust logic to extend and shrink a wire when the node it is connected to is on the same line as the wire itself.
closed 2025-08-21 by ra3xdh