ra3xdh/qucs_s #1453: QT signal delivery delay when editing a component

funnydog, 2025-08-26:
Hi,

when I edit the properties of a component using the `Edit Properties` context-menu command like below, the changes I make are not properly saved when the dialog is closed with the keyboard.

For example, in the following dialog:

[image]

if I change the capacitance value and then hit Enter (or use the shortcut Alt+O), the new value is not saved and the schematics keeps showing the old capacitance.

This never occurs when clicking on the OK button with the mouse. Besides it isn't 100% reproducible with the Enter key but it's always reproducible with the Alt+O shortcut.

I'm using the Arch Linux package (25.1.2) but it's also reproducible with the latest git head (25.2).

As I understand, this is due to a delay in the delivery of the focusOut signal in the QT library: the accept signal for QButtonBox is delivered before the focusOut signal  of the propertyTable cell.

I have to say I'm not a QT expert but it seems this is a known issue and people work around it by setting the focus on the propertyTable before saving back the changed contents.

Therefore I tried to patch the code in `qucs/components/componentdialog.cpp` as following:

```c++
void ComponentDialog::slotApplyButton()
{
...
    // ensure the table has focus to avoid data synchronization issues
    // with the focusOut signal delivery delay
    propertyTable->setFocus();
    row = 0;
    for (Property* property : component->Props)
      if ((property->simulators & QucsSettings.DefaultSimulator) != QucsSettings.DefaultSimulator) {
        continue;
      }

      // Ignore sweep parameters here.
      if (hasSweep && sweepParamWidget.contains(property->Name))
        continue;
...
```

And this solved the issue above. I'm not sure it's the right fix but it seems to work for me and solves the very annoying problem of having to change the same component twice or thrice.

funnydog, 2025-08-26:
Hello, thank you for your response.

As I suspected the problem is difficult to reproduce.

I think I will do further research before I submit code that may be not needed at all.

funnydog, 2025-08-28:
After further research I think the problem is related to the use of QTableWidget::openPersistentEditor(): in ComponentDialog::updatePropertyTable().

That method is called to let the user update the value of the cell if the property isn't special (options and files).

In the QT forum, experienced devs seem to discourage the use of openPersistentEditor() (at least for performance reasons) but they don't give a clear reason ([link](https://forum.qt.io/post/720466) here).

Some of them say it's better to use a QLineEdit widget, following the way it's done for options (that use QComboBox).

```c++
void ComponentDialog::updatePropertyTable(const Component* updateComponent)
{
...
      // Add text edit if no options found.
      else
      {
	      QLineEdit *edit = new QLineEdit(property->Value);
	      propertyTable->setCellWidget(row, 1, edit);
        propertyTable->setItem(row, 1, new QTableWidgetItem("", TextEditCell));
        // propertyTable->openPersistentEditor(propertyTable->item(row, 1));
        // propertyTable->item(row, 1)->setText(property->Value);
      }    

```

Then I made the corresponding change in the code that reads the text back from the widget in ComponentDialog::slotApplyButton().

```c++
void ComponentDialog::slotApplyButton()
{
...
        else
        {
	        QLineEdit *edit = static_cast<QLineEdit*>(propertyTable->cellWidget(row, 1));
	        property->Value = edit->text();
          // property->Value = propertyTable->item(row, 1)->text();
        }
```

And this solved the issue for me.

I don't know why this occurs and I'm not experienced enough with the QT internals to tell why openPersistentEditor() behaves the way it does. But the QLineEdit widget behaves correctly, maybe it's been tested/debugged more.

PS: I can reproduce this issue 100% of the time and it could be related to my system configuration:
* QT6 version 6.9.1
* X11 windowing system with XFCE 4.20 with the compositor disabled
* NVidia proprietary driver version 470.256.02

ra3xdh, 2025-09-02:
Fixed by #1459